madsci1016/Arduino-PS2X #10: ps2 button problem

moxitpatelxolo, 2017-07-09:
while upload this code,
when i press one button then it will print two button

ari86m, 2017-07-09:
I assume you are saying that when you press one button, the software reports another button has also been pressed. We have seen this behavior when using an off brand controller wired into an Arduino. It looks like there is some noise during communication. We don't see any error detection information in the data, so you would have to filter the data in software or do something to resolve the presumed noise problem. We saw some patterns in the incorrect packets received by an Arduino. Here are some things that could be done to prevent this. 

Band-aid - Filter packet in the software:
1- discard any packet that contains more than 1 function (buttons & joysticks)
2- only keep packets that contain individual functions & combinations that should be allowed.
3- discard packets that match the finger prints of known incorrect packets.

Permanent fix in the hardware: 
1- someone with an oscilloscope could see what the data signal really looked like at various locations in the hardware path of the signal & an "enthusiast" or engineer might be able to remedy the cause of the noise. 

Good Luck!

goatfanboi23, 2023-11-18:
I had a very similar problem when using an Arduino mega instead of an Uno (yes, I was using a cheap controller). I solved this problem by adding a delay of 10ms in the `loop` function. This completely eliminated buttons being registered twice and I have not had the problem since. I am not 100% sure why this works my thought is that maybe the increased clock frequency 8mhz (uno) to 16mhz (mega) lead to the library registering buttons twice. Hopes this helps anyone with this problem.