madsci1016/Arduino-PS2X #19: Example works with Uno but not with DUE

dsyleixa, 2019-04-04:
the lib seems to be bugged or outdated for ARM CORTEX Boards (Due and different ones like Adafruit M4):

```
compile errors:
Arduino: 1.8.8 (Windows 7), Board: "Arduino Due (Programming Port)"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X239A_0X8022 -ide-version=10808 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_721665 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose D:\arduino\portable\sketchbook\libraries\PS2X_lib\examples\PS2X_Example\PS2X_Example.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X239A_0X8022 -ide-version=10808 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_721665 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose D:\arduino\portable\sketchbook\libraries\PS2X_lib\examples\PS2X_Example\PS2X_Example.ino
Using board 'arduino_due_x_dbg' from platform in folder: D:\arduino\portable\packages\arduino\hardware\sam\1.6.11
Using core 'arduino' from platform in folder: D:\arduino\portable\packages\arduino\hardware\sam\1.6.11
WARNUNG: Kategorie 'Buttons' in der Bibliothek ButtonClass ist ungültig und wird auf 'Uncategorized' festgelegt
WARNUNG: Kategorie 'TFT menu' in der Bibliothek MenuClass ist ungültig und wird auf 'Uncategorized' festgelegt
WARNUNG: Kategorie '' in der Bibliothek XPT2046 ist ungültig und wird auf 'Uncategorized' festgelegt
Build-Optionen wurden verändert, alles wird neu kompiliert
Detecting libraries used...
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10808 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\variants\\arduino_due_x" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_721665\\sketch\\PS2X_Example.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10808 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\variants\\arduino_due_x" "-ID:\\arduino\\portable\\sketchbook\\libraries\\PS2X_lib" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_721665\\sketch\\PS2X_Example.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10808 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\system/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.11\\variants\\arduino_due_x" "-ID:\\arduino\\portable\\sketchbook\\libraries\\PS2X_lib" "D:\\arduino\\portable\\sketchbook\\libraries\\PS2X_lib\\PS2X_lib.cpp" -o nul
D:\arduino\portable\sketchbook\libraries\PS2X_lib\PS2X_lib.cpp:5:20: fatal error: avr/io.h: No such file or directory

 #include <avr/io.h>

                    ^

compilation terminated.

Bibliothek PS2X_lib im Ordner: D:\arduino\portable\sketchbook\libraries\PS2X_lib (legacy) wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).
```

would you please kindly fix that?

dsyleixa, 2019-04-11:
is this Repo no longer maintained?

dsyleixa, 2020-05-31:
no hope that someone will re-write this lib to work also with SAM, SAMD, and hopefully also ESP MCUs?

dsyleixa, 2020-05-31:
errors for SAMD51:
```
Arduino: 1.8.9 (Windows 7), Board: "Adafruit Feather M4 Express (SAMD51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_651464 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS-Atmel.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.openocd.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.CMSIS.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0.path=D:\arduino\portable\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose D:\Arduino\portable\sketchbook\libraries\PS2X_lib\examples\PS2X_Example\PS2X_Example.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_651464 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS-Atmel.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.openocd.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.CMSIS.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0.path=D:\arduino\portable\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose D:\Arduino\portable\sketchbook\libraries\PS2X_lib\examples\PS2X_Example\PS2X_Example.ino
Using board 'adafruit_feather_m4' from platform in folder: D:\arduino\portable\packages\adafruit\hardware\samd\1.5.8
Using core 'arduino' from platform in folder: D:\arduino\portable\packages\adafruit\hardware\samd\1.5.8
Detecting libraries used...
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"PS2X_Example.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10809 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.8\\cores\\arduino/TinyUSB" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.8\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.8\\cores\\arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.8\\cores\\arduino/TinyUSB" "-ID:\\arduino\\portable\\packages\\adafruit\\hardwar [... cut ...]
D:\arduino\portable\sketchbook\libraries\PS2X_lib\PS2X_lib.cpp:204:20: note: in expansion of macro 'portOutputRegister'

   _att_lport_clr = portOutputRegister(lport) + 1;

                    ^~~~~~~~~~~~~~~~~~

Mehrere Bibliotheken wurden für "PS2X_lib.h" gefunden
 Benutzt: D:\arduino\portable\sketchbook\libraries\PS2X_lib
 Nicht benutzt: D:\arduino\portable\sketchbook\libraries\PS2X_lib_Dexter
Bibliothek PS2X_lib im Ordner: D:\arduino\portable\sketchbook\libraries\PS2X_lib (legacy) wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit Feather M4 Express (SAMD51).
```

dsyleixa, 2020-05-31:
Due (errors for SAM3X8E):

```
Arduino: 1.8.9 (Windows 7), Board: "Arduino Due (Programming Port)"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=arduino:sam:arduino_due_x_dbg -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_651464 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -verbose D:\Arduino\portable\sketchbook\libraries\PS2X_lib\examples\PS2X_Example\PS2X_Example.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=arduino:sam:arduino_due_x_dbg -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_651464 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=D:\arduino\portable\packages\arduino\tools\bossac\1.6.1-arduino -verbose D:\Arduino\portable\sketchbook\libraries\PS2X_lib\examples\PS2X_Example\PS2X_Example.ino
Using board 'arduino_due_x_dbg' from platform in folder: D:\arduino\portable\packages\arduino\hardware\sam\1.6.12
Using core 'arduino' from platform in folder: D:\arduino\portable\packages\arduino\hardware\sam\1.6.12
Build-Optionen wurden verändert, alles wird neu kompiliert
Detecting libraries used...
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10809 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_651464\\sketch\\PS2X_Example.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10809 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "-ID:\\arduino\\portable\\sketchbook\\libraries\\PS2X_lib" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_651464\\sketch\\PS2X_Example.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10809 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\ardui [... cut ...]
D:\arduino\portable\sketchbook\libraries\PS2X_lib\PS2X_lib.cpp:204:20: note: in expansion of macro 'portOutputRegister'

   _att_lport_clr = portOutputRegister(lport) + 1;

                    ^

Mehrere Bibliotheken wurden für "PS2X_lib.h" gefunden
 Benutzt: D:\arduino\portable\sketchbook\libraries\PS2X_lib
 Nicht benutzt: D:\arduino\portable\sketchbook\libraries\PS2X_lib_Dexter
Bibliothek PS2X_lib im Ordner: D:\arduino\portable\sketchbook\libraries\PS2X_lib (legacy) wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Arduino Due (Programming Port).
```

plainchn, 2021-11-25:
did you solved?

dsyleixa, 2021-11-25:
not solved yet.

plainchn, 2021-11-25:
line190:
 ​  ​uint32_t​            lport;                   ​//
 Port number for this pin 

revise:
 ​Pio*​            lport;                   ​//
 Port number for this pin 

The example complied succeed,but  serial print noticed that can't find controller