madsci1016/Arduino-PS2X #26: Arduino PS2X module not working with PS1 controller

Digital-Rookie4, 2020-05-14:
I have been trying to use the PS2X module to try to use my PS1 remote with my arduino (I am guessing that PS1 controllers will work because they are virtually the same ). The problem is that I don't want to download the module from the original website because it is not secure (it had HTTP instead of HTTPS) and the download file could contain viruses. Instead I got it off the arduino website. I have wired my controller correctly and have triple checked that all the wires are in the right positions and they seem to be. I then used the arduino example program given by the safely downloaded file. I complied it onto my arduino and I got the following message in the serial `No controller found, check wiring, see readme.txt to enable debug. visit www.billporter.info for troubleshooting tips
Unknown Controller type.` I then looked into the text file to debug it and it told me to uncomment 2 of the lines which is what I did. I then ran the example program file again like how the readme file told me to, and I still get the same errors.

How its supposed to be wired:

[image]

How I have wired it:

[image]

maxbambucha, 2020-05-14:
Hi
As per Bill Porter' troubleshooting guide, try adding a 10k resistor between DATA (brown wire, or pin 1 on the controller port) and VCC (try both 3.3V and 5V in that order) on the Arduino
That helped me.
--
Отправлено из Mail.ru для Android четверг,

Digital-Rookie4, 2020-05-14:
@maxbambucha  no luck :(

Digital-Rookie4, 2020-05-14:
@maxbambucha  could it be because I am using a PS1 controller and not a PS2 controller

maxbambucha, 2020-05-14:
For me it worked ok with original PS1 controller as well as Chinese wireless Dualshock 2.
--
Отправлено из Mail.ru для Android четверг,

Digital-Rookie4, 2020-05-14:
Do you know at all possibly why the debugger is not working?

Digital-Rookie4, 2020-05-14:
@maxbambucha  Do you know at all possibly why the debugger is not working?

Digital-Rookie4, 2020-05-14:
@maxbambucha   Right ok firstly I read something wrong that you wrote you said put a 10K resistor through the data wire and the **VOLTAGE SUPPLY**. Ok, I never did that, I only did the first part. Can you potentially show me how you would wire this.

SerEpicPanda, 2020-05-14:
I think that the PS1 controller will give a different ID when the program checks what type controller it is. You may still get data out if you remove the type check. I also had a similar problem when using a wireless PS2 controller and I added a call to reconfigure the controller 500ms after the initialisation.

Digital-Rookie4, 2020-05-14:
@SerEpicPanda  I have tried removing the checker before, when I did this and pressed on a button it referred to a different button on the controller or even 2 buttons.

SerEpicPanda, 2020-05-14:
So the controller is working but you will have to get the raw data from it to find out which data bits are for each button. PS2 controllers have analog sticks and analog buttons so they give more data to the arduino than a PS1 controller will. It's likely that the data bits are just different to what the library is expecting.

SerEpicPanda, 2020-05-15:
I also found this link 
https://theteamasterix.wordpress.com/robo-sprint-2/robo-sprint/

It's made by someone who got the PS1 controller working with this library but they say it came up as a guitar hero controller

maxbambucha, 2020-05-15:
This is what I ment by adding 10K resistor between data pin and vcc.
Also I agree that sometimes controller needs to be initialized more than once, so i added a loop of 10 appempts until the answer is recieved or maximum attempts reached.
--
Отправлено из Mail.ru для Android пятница,

Digital-Rookie4, 2020-05-15:
@SerEpicPanda  I tired that but it did not work, I still got the same error as before. This is how I wired it:
[image].

Would you possibly suggest to just edit the library so that it works with each button

If thats the case, I have a problem, how am I going to reprogram the joysticks if there is only 2 and if they are already connected up to another button in the library

Digital-Rookie4, 2020-05-15:
@maxbambucha  How would I exactly initialise it 10 times

Digital-Rookie4, 2020-05-15:
@SerEpicPanda  @maxbambucha  is L1 or R1 the joysticks?

SerEpicPanda, 2020-05-15:
I think the circuit looks fine. It may be that the power supply isn't good enough but I doubt it.

To initialise it multiple times you can just call the initialise line multiple times or call the reconfig line several times. Sometimes the controller just needs time to startup which is why there is a delay at the beginning.

I'm also not sure what you mean by 'is R1 or L1 the joysticks' as they are both shoulder buttons. The original PS1 controller didn't have joysticks but then they released the Dualshock 1 which had two. The click button of each joystick is R3 and L3.

Digital-Rookie4, 2020-05-16:
@SerEpicPanda  Right, I have tried using 5V instead of 3.3V and the arduino now thinks it is a guitar hero controller. This message came up `Controller refusing to enter Pressures mode, may not support it. 
GuitarHero Controller Found
` This PS1 controller does have joysticks:
[image]

SerEpicPanda, 2020-05-16:
The dualshock 1 does not support pressures so that error is correct. Are you able to get the joystick values out when it thinks it's a guitar hero Contoller.

Digital-Rookie4, 2020-05-16:
@SerEpicPanda  One thing I did recognise is that if I click the square button I get `orange fret pressed` along with this I also get a message saying `wammy bar position:` if I then move the joystick while holding the square button I get the value of the joystick. I also recognise that if you hold the square button you get `orange fret pressed` once then you get lots of `wammy bar position` due to holding down the button, if I move the joystick along with this the `wammy bar position:` changes

SdWydler, 2020-05-16:
Hello  this is a guide with diff variables I used from lynmotion .  I hope this helps  http://www.lynxmotion.com/images/jpg/PS2-V1-Transmitter.png

Digital-Rookie4, 2020-05-16:
@SdWydler  its a picture though.

SdWydler, 2020-05-16:
http://www.lynxmotion.com/driver.aspx?Topic=ps2-versions

SdWydler, 2020-05-16:
lol

Digital-Rookie4, 2020-05-16:
@SdWydler  One thing though. I am using a PS1 controller and not PS2