madsci1016/Arduino-PS2X #37: Joysticks ohne Funktion ?

DK0975, 2023-08-30:
Hallo. Ich habe einen PS2 Wireless Controller genauso angeschlossen wie es für das Beispiel erforderlich ist. Alles mehrfach kontrolliert. Im seriellen Monitor kommt bei allen Tasten eine Ausgabebestätigung. Beim drücken der Joysticks kommt die Meldung da R3, bzw. L3 gedrückt wurde. Beim betätigen der Joysticks in X oder Y (egal ob rechts oder links) kommt im seriellen Monitor keine Ausgabe. Woran könnte das liegen ?

SerEpicPanda, 2023-08-31:
Just checking that you are holding R1 or L1 to print the values as is in the example?

DK0975, 2023-08-31:
Beim drücken von L1 oder R1 bekomme ich Werte (weiß aber jetzt nicht welche).

SerEpicPanda, 2023-08-31:
In this bit of code it shows which values are printed whilst R1 or L1 are pressed. In this case LY, LX, RY, RX. you could add labels in the print statements to make it clearer.

`if(ps2x.Button(PSB_L1) || ps2x.Button(PSB_R1)) { //print stick values if either is TRUE
      Serial.print("Stick Values:");
      Serial.print(ps2x.Analog(PSS_LY), DEC); //Left stick, Y axis. Other options: LX, RY, RX  
      Serial.print(",");
      Serial.print(ps2x.Analog(PSS_LX), DEC); 
      Serial.print(",");
      Serial.print(ps2x.Analog(PSS_RY), DEC); 
      Serial.print(",");
      Serial.println(ps2x.Analog(PSS_RX), DEC); 
    }    `